ros/xacro #292: Slow speed when generating a URDF

captain-yoshi, 2021-10-08:
It takes 40 seconds for my pc (_Intel i7-7700HQ 2.8GHz_) to generate a URDF containing links and joints that represent a collection of collision objects (10 shelves on top of each other). Information such as subframes is also added. There is a lot of xacro loops used. The URDF generated has a size of 500kB.

Is it normal for this process to be slow ? My xacro skills may be to blame, could xacro `for loops` introduces some sort of slowdown that could be improved with #291 ?

You can run the example ([xacro_scene.zip](https://github.com/ros/xacro/files/7313195/xacro_scene.zip))
```bash
rosrun xacro xacro scene.urdf.xacro > scene.urdf
```
<p align="middle">
[image]
</p>

rhaschke, 2021-10-08:
Wow! What a huge shelf! Unfortunately, I have to blame python's `xml.dom.minidom`. I filed a PR in the past to improve the situation for Python 2.7, but they didn't accept it (not trying to risk stability): https://bugs.python.org/issue24424
Maybe the patch provided there helps you as well.

For Python 3.x, the situation is slightly better. We typically generate our URDFs once when building. 
To this end xacro provides some [convience cmake functions](https://github.com/ros/xacro/blob/noetic-devel/cmake/xacro-extras.cmake).

captain-yoshi, 2021-10-18:
Comparing timing on same machine (i7-2600) gives roughly the same table:
|          | Melodic | Noetic |
|----------|---------|--------|
| 1 Shelf  | 22 sec  | 1 sec  |
| 10 Shelf | **54 min**  | 35 sec |

@rhaschke Trying to generate a urdf from xacro for Python 3.x (ROS Noetic) using the conveniance cmake functions. Do you have an exemple on how to do that ? Not sure if I have to uncomment parts of cmake...

captain-yoshi, 2022-03-28:
Benchmarked my Pisa tower example using a loop and an unrolled loop on Noetic  (see below). It seems that looping can be quite taxing. Maybe this can be improved with #291.

| Test     | Time (s) |
|---------------|:--------:|
| Loop          |    28    |
| Unrolled Loop |     8    |

```xml
   <!-- Unrolled loop x10 -->

  <xacro:shelf name="bookshelf1"
               frame_id="world"
               xyz="0.8 0 0"
               h_board_thickness="0.02"/>

  <xacro:shelf name="bookshelf2"
               frame_id="world"
               xyz="0.8 0 1"
               h_board_thickness="0.02"/>
 ...

  <xacro:shelf name="bookshelf10"
               frame_id="world"
               xyz="0.8 0 9"
               h_board_thickness="0.02"/>
```