ros/xacro #298: cmake xacro_add_files fails with xacro.load_yaml() instead of load_yaml()

nullket, 2021-11-05:
Hello,

I have one package (`urdf_package`) holding an URDF including xacro properties. I try to use some static values and parse the file directly with `urdf::Model model` in c++ within an other package (`package_b`).  

When I was using the cmake function `xacro_add_files()` it worked for the first time, but I got a warning that the macro `load_yaml()` i was using in my urdf was deprecated and instead `xacro.load_yaml()` should be used. Alright I changed that bit of code and it works just fine on my computer (like in the old saying). But our [industrial ci](https://github.com/ros-industrial/industrial_ci) fails during the cmake function call `xacro_add_files()` on [Line 77](https://github.com/ros/xacro/blob/47f3605ab06a9e6a7b00f2873d3f2e5b98dde6b7/cmake/xacro-extras.cmake#L77) when I use  `xacro.load_yaml()`  instead of  `load_yaml()`.

I can not figure out why. Any ideas?

For simplification I hard-coded some paths but the concept is:  
**cmake of package_b (not the urdf package)**
```cmake
xacro_add_files(${CMAKE_CURRENT_SOURCE_DIR}/../urdf_package/urdf/robot.xacro TARGET media_files REMAP ROBOT:=robot_1)
```

**ci build log**
```log
Errors << package_b:make /root/target_ws/logs/package_b/build.make.000.log
name 'xacro' is not defined 
when evaluating expression 'xacro.load_yaml('$(find urdf_description)/robots/$(arg ROBOT)/config/robot.yaml')' 
when evaluating expression 'robot['maps']['map_ENU_joint']['x']'
when processing file: /root/target_ws/src/ros/src/package_b/../urdf_package/urdf/robot.xacro 
```
**Beginning of URDF**
```xml
<?xml version="1.0"?>
<!-- Revolute-Revolute Manipulator -->
<robot name="$(arg ROBOT)" xmlns:xacro="http://ros.org/wiki/xacro">
<xacro:property name="robot" value="${xacro.load_yaml('$(find urdf_description)/robots/$(arg ROBOT)/config/robot.yaml')}"/>
....
```

rhaschke, 2021-11-05:
When it works locally for you, probably industrial_ci uses a different (older) version of xacro?
Can you check the used versions?
Did `industrial_ci` issue the `xacro.load_yaml` deprecation warning as well? If not, this hints at different versions.

nullket, 2021-11-09:
You are perfectly right. It was not the fault of `industrial_ci` but my docker build script which is used to to create the correct environment did not always pull the latest ros base image first (it simply used the local available and thus old version). Thanks for the quick response which lead me to the solution!

closed 2021-11-09 by nullket