ros/xacro #301: Yaml dot resolution within xacro call

Doomerdinger, 2021-12-09:
Normally if I were to load a yaml file
```yaml
---
testbool: false
```
I would be able to load that up with `<xacro:property name="bar" value="${xacro.load_yaml('file')}"/>`
and then access `testbool` by doing `${bar.testbool}`.
However, if the xacro was called using `xacro:call` (at least that seems to be the cause as best I can tell), it appears that this dot functionality no longer works.
I am forced to use `${bar['testbool']}`.

rhaschke, 2021-12-09:
Could you provide a complete minimal (failing) example, please?

Doomerdinger, 2021-12-10:
I tried crafting a simple example from scratch, but the issue is not quite what I thought so it doesn't reproduce there.

I'll try cutting my real world example down to size, but that will take a bit of time.

Note: Using `bar['testbool']` does work for me even in the real world example, it is something relating to the dot accessor. 

I'll attach a zip file of my simple scenario that I thought would reproduce the issue but does 
not, just in case you want to see what I was trying to explain.
[test_xacro.zip](https://github.com/ros/xacro/files/7689177/test_xacro.zip)

Doomerdinger, 2021-12-10:
I believe I was wholly mistaken about the source of my issue. I was staring at it for ages and it just slipped through the cracks.

I ended up passing in the result of a yaml read into `xacro` (like `xacro xacro file.xacro yaml_arg:="{'foobar': True}"`) and _that_ allows the argument `yaml_arg` to be accessed with [] but not with the dot operator.

I don't believe this is necessarily a bug. If you can confirm it is not, close this and I shall work around it. If you would like more information I can throw something together.

rhaschke, 2021-12-10:
I don't think you can pass in a `dict` object via an argument as described, because `arg` results are expected to be strings. :confused: 
In any case, you can use `xacro.dotify()` to turn a standard `dict` object into a dotified `dict`:
```xml
<a xmlns:xacro="http://www.ros.org/wiki/xacro" xmlns:a="http://www.ros.org/a">
	<xacro:property name="d" value="${dict(foo=False)}" />
	<xacro:property name="dotified" value="${xacro.dotify(d)}" />
	${d['foo']} ${dotified.foo}
</a>
```

closed 2021-12-10 by rhaschke

Doomerdinger, 2021-12-10:
Passing in the argument as I described does actually seem to work, I just can't use the dot operator on it. 

Doing as you have suggested and using `xacro.dotify` seems to allow me to use the dot operator on it no problem. Thank you very much!