ros/xacro #302: Cannot run xacro via ros2 run in Windows

wmmc88, 2021-12-26:
```
ros2 run xacro xacro
Traceback (most recent call last):
  File "C:\dev\ros2_rolling\install\Scripts\ros2-script.py", line 33, in <module>
    sys.exit(load_entry_point('ros2cli==0.15.0', 'console_scripts', 'ros2')())
  File "C:\dev\ros2_rolling\install\Lib\site-packages\ros2cli\cli.py", line 89, in main
    rc = extension.main(parser=parser, args=args)
  File "C:\dev\ros2_rolling\install\Lib\site-packages\ros2run\command\run.py", line 70, in main
    return run_executable(path=path, argv=args.argv, prefix=prefix)
  File "C:\dev\ros2_rolling\install\Lib\site-packages\ros2run\api\__init__.py", line 64, in run_executable
    process = subprocess.Popen(cmd)
  File "c:\python38\lib\subprocess.py", line 858, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "c:\python38\lib\subprocess.py", line 1311, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
OSError: [WinError 193] %1 is not a valid Win32 application
```

this seems like the same error that happened with some other ros2 python programs: https://github.com/ros-visualization/rqt_console/issues/20

I think similar fixes there need to happen to xacro to be able to run this package under windows.

Directly invoking `xacro` also doesnt work since shebangs arent handled by windows.

rhaschke, 2021-12-27:
Thanks for reporting the issue. As I don't have a Windows installation available, please apply the suggested changes yourself, validate, and file a PR.

rhaschke, 2022-10-22:
Fixed via #304.

PR #304 (2022-10-21): install xacro using console_scripts entrypoint
This changes xacro to be install using a console_script entrypoint, which now fixes running `xacro` and `ros2 run xacro xacro` in windows. linux functionality should be unchanged. 

closes #302 
supersedes #303
closed 2022-10-22 by rhaschke