ros/xacro #305: Include macro with namespace propagates across all subsequent includes

captain-yoshi, 2022-01-19:
Adding the `include` macro with a namespace propagates the namespace across all subsequent `include` macros. I would expect the namespace to not be propagated for other includes within the include with a namespace.

```bash
$ rosrun xacro xacro scene.urdf.xacro > scene.urdf

unknown macro name: xacro:framebox
when instantiating macro: object (./car.urdf.xacro)
in file: scene.urdf.xacro
```
[example.zip](https://github.com/ros/xacro/files/7899400/example.zip)
### Example
scene.urdf.xacro
```xml
<?xml version="1.0" ?>
<robot name="planning_scene" xmlns:xacro="http://www.ros.org/wiki/xacro">
  <xacro:include filename="./car.urdf.xacro" ns="car"/>
 
  <xacro:car.object root_id="root"/>
 
</robot>
```

car.urdf.xacro
```xml
<?xml version="1.0" ?>
<robot name="collision_object" xmlns:xacro="http://www.ros.org/wiki/xacro">
  <xacro:include filename="./utils.urdf.xacro"/>

  <xacro:macro name="object" params="prefix:='' root_id:='root'">

    <!-- Root link -->
    <link name="${prefix}${root_id}"/>

    <!-- Changing framebox to car.framebox works -->
    <xacro:framebox name="${prefix}car"
                    dimensions="1 2 3"
                    frame_id="${prefix}${root_id}"/>
  </xacro:macro>

</robot>
```
utils.urdf.xacro
```xml
<?xml version="1.0" ?>
<robot name="utilities" xmlns:xacro="http://www.ros.org/wiki/xacro">
 
  <!-- Store tf information as a joint for the collision object -->
  <xacro:macro name="tf" params="name parent *origin">
    <joint name="${name}" type="fixed">
      <xacro:insert_block name="origin" />
      <parent link="${parent}" />
      <child link="${name}" />
      <axis xyz="0 0 1" />
    </joint>
  </xacro:macro>
 
  <!-- Add primitive box -->
  <xacro:macro name="framebox" params="name frame_id:='' dimensions xyz:='0 0 0' rpy:='0 0 0'">
    <!-- Parent joint -->
    <xacro:if value="${frame_id != ''}">

      <!-- Changing tf with car.tf works -->
      <xacro:tf name="${name}" parent="${frame_id}">
        <!-- Add fixed origin and include z dimension -->
        <origin xyz="${xyz.split()[0]} ${xyz.split()[1]} ${float(xyz.split()[2]) + float(dimensions.split()[2])}"/>
      </xacro:tf>
    </xacro:if>
    
    <!-- Child link -->
    <link name="${name}">
      <collision>
        <!-- Replace collision visualization because box origin is in the middle -->
        <origin xyz="0 0 ${-float(dimensions.split()[2])/2}" rpy="${rpy}"/>
        <geometry>
          <box size="${dimensions}" />
        </geometry>
      </collision>
    </link>
 
  </xacro:macro>
 
</robot>
```

rhaschke, 2022-01-19:
Your expectation is correct. This bug was recently fixed (#297) but not yet released.

captain-yoshi, 2022-01-19:
Oh it's fixed on the melodic branch. I'm on noetic branch, that's why I thought it was a bug. Thx for the fix !

captain-yoshi, 2022-01-19:
@rhaschke The fix seems to propagate through a `xacro:property` loaded from a yaml.

My real world example include the file below (with a namespace). The bowl property must be called which the namespace `ns.bowl`
```xml
<?xml version="1.0" ?>
<robot name="scene" xmlns:xacro="http://www.ros.org/wiki/xacro">

  <xacro:include filename="$(find moveit_benchmark_suite_resources)/objects/metadata.xacro" />

  <xacro:macro name="object" params="prefix:='' root_id:='root'">

    <!-- Object Root -->
    <link name="${prefix}${root_id}"/>
    <xacro:framebox name="${prefix}boxlid"
                    dimensions="${bowl['bb'[0] 0 0]}"
                    frame_id="${prefix}${root_id}"/>

```
metadata.xacro
```xml
<?xml version="1.0" ?>
<robot xmlns:xacro="http://www.ros.org/wiki/xacro" name="panda">

  <xacro:property name="pkg_path"   value="$(find moveit_benchmark_suite_resources)/objects/"/>
 
  <!-- Load metadata of all meshes -->
  <xacro:property name="bowl"       value="${load_yaml(pkg_path + 'bowl.yaml')['bowl']}"/>
 
</robot>

```

rhaschke, 2022-01-19:
I don't yet understand whether there is still an open issue or not. Could you please provide a full minimal example again?

captain-yoshi, 2022-01-19:
Here is a **new** simplified example that does not work with latest commit from melodic-devel branch. **FYI** The original example works.

[new_example.zip](https://github.com/ros/xacro/files/7900328/new_example.zip)

rhaschke, 2022-01-19:
Thanks for the clarification. Indeed, there was still a bug. Should be fixed via #306. Could you please verify on your real-world files?

captain-yoshi, 2022-01-20:
Wow that was fast ! The bug is fixed.

rhaschke, 2022-01-20:
Fixed via #306.

closed 2022-01-20 by rhaschke