ros/xacro #307: Macro arg resolution with namespaces

Doomerdinger, 2022-01-31:
After the 1.13.15 update I encounter issues when using macros imported with namespaces.

generic_utils.xacro
```xml
<xacro:macro name="do_a_thing" params="first_arg">
  <xacro:if value="${first_arg}">
    Foobar
  </xacro:if>
</xacro:marco>
```

```xml
<xacro:macro name="my_macro" params="dict_config">
  <xacro:include filename="$(find example_pack)/folder/generic_utils.xacro" ns="generic_utils"/>
  
  <xacro:generic_utils.do_a_thing first_arg="${dict_config.item}"/>
  <xacro:generic_utils.do_a_thing first_arg="${dict_config['item']}"/>
</xacro:maro>
```
I've only tested with a dict input myself (using load yaml), I cannot say if this issue persists elsewhere.

The above setup will result in an error, complaining that `dict_config` is not defined. This will come from within the generic_utils.xacro file.
If the namespace on the `include` is omitted so it is just `<xacro:include filename="$(find example_pack)/folder/generic_utils.xacro"/>` and subsequent references to the namespace are removed, it seems to work fine.

I assume this is something to do with the changes either on #306 or #297.

rhaschke, 2022-01-31:
Thanks for reporting this issue. Indeed, this was a consequence of the stricter namespace handling introduced recently.
#308 should fix this. Could you please verify?

Doomerdinger, 2022-02-01:
Thanks for the quick update, I'll test this in the next day or two.

Doomerdinger, 2022-02-03:
While it seems that this (may) solve the initial issue cited, it appears to break the set property value in parent functionality, eg
```xml
<xacro:property name="parent_prop" scope="parent" value="whatever"/>
```

rhaschke, 2022-02-03:
In which context does this fail now?

Doomerdinger, 2022-02-03:
If I have a parent calling a macro, and that macro does the above `<xacro:property name="parent_prop" scope="parent" value="whatever"/>` and the parent expects that property to be set after the call, it is not set.

rhaschke, 2022-02-03:
But there is a non-failing unit test for this:
https://github.com/ros/xacro/blob/272932713d60695d2ab92e214827b22cf3c97cbe/test/test_xacro.py#L495-L501

Doomerdinger, 2022-02-03:
Interesting. I will test further.

Doomerdinger, 2022-02-03:
Ok I think I have it.

If the parent includes the macro with a namespace and then the parent calls that macro, properties that are set to the parent scope must be accessed with the namespace accessor. This is not how it worked before, but I am not sure if it is a bug or not.

Would make sense that setting something in the parent namespace would put the property in the namespace of whatever the _parent_ is in, not whatever my current namespace is.

`macro_utils.xacro`
```xml
<xacro:macro name="set_property" params="">
  <xacro:property name="parent_prop" scope="parent" value="whatever"/>
</xacro:macro>
```
`parent.xacro`
```xml

<xacro:include filename="$(find utils)/urdf/macro_utils.xacro" ns="macro_utils"/>
<xacro:macro_utils.set_property/>
<a foo="${parent_prop}"/></a> <!-- ERROR -->
<a foo="${macro_utils.parent_prop}"/></a> <!-- A-OK -->

```

rhaschke, 2022-02-04:
Interesting example. In this case, essentially, _two_ new scopes are created:
- `root` scope
   - namespaced `include` scope
      - `macro` scope

Thus, defining a property within the root scope from within the namespaced macro requires lifting the property by two scopes, which is not yet supported. Actually, We could even have multiple nested namespaced includes, which would create even more scopes. 
Setting a property within the parent scope may occur in two contexts:
1. From within a macro. In that case, one wants to set the property in the caller's scope.
2. From within the included file. In that case, one wants to set the property in the includer's scope.

I will adapt the PR appropriately.

rhaschke, 2022-02-12:
Fixed via #308

PR #308 (2022-02-12): Fix property resolution within namespaces
Need to store current symbol table as parent when creating scoped namespace
Fixes #307
closed 2022-02-12 by rhaschke