ros/xacro #309: Xacro evaluates commented out code

alaurenzi, 2022-02-01:
After a binary update, I am facing a bug where xacro will evaluate code sections that are commented out.

## How to reproduce
 - Create a minimal xacro file, save it as  'text.xacro`
```xml
<robot xmlns:xacro="http://www.ros.org/wiki/xacro">
<!-- $(find dontexist) -->
</robot>
```

 - Run `xacro test.xacro`
 - Output:
```
arturo@arturo-iit-desktop ~/code/xacro_bug $ xacro test.xacro 
resource not found: dontexist
ROS path [0]=/opt/ros/melodic/share/ros
ROS path [1]=/home/arturo/code/robot_ws/src
ROS path [2]=/home/arturo/code/robot_ws/install/share
ROS path [3]=/home/arturo/code/robot_ws/install/lib
ROS path [4]=/opt/ros/melodic/share
ROS path [5]=/opt/xbot/share
ROS path [6]=/opt/xbot/lib
when processing file: test.xacro
```

## Some info about my setup
```
arturo@arturo-iit-desktop ~/code/xacro_bug $ which xacro
/opt/ros/melodic/bin/xacro

arturo@arturo-iit-desktop ~/code/xacro_bug $ dpkg -S /opt/ros/melodic/bin/xacro
ros-melodic-xacro: /opt/ros/melodic/bin/xacro

arturo@arturo-iit-desktop ~/code/xacro_bug $ apt-cache show ros-melodic-xacro
Package: ros-melodic-xacro
Version: 1.13.15-1bionic.20220120.162849
Architecture: amd64
Maintainer: Robert Haschke <rhaschke@techfak.uni-bielefeld.de>
Installed-Size: 172
Depends: ros-melodic-roslaunch
Homepage: http://ros.org/wiki/xacro
Priority: optional
Section: misc
Filename: pool/main/r/ros-melodic-xacro/ros-melodic-xacro_1.13.15-1bionic.20220120.162849_amd64.deb
Size: 48392
SHA256: 94b072fadc0e09c8fb153b9a7bd27c0d85454e034122116556acc334d0012a06
SHA1: ea8747bc28bec20a7d3b380b6c714c3f2dc0d7e9
MD5sum: dc1b6217b004853d052926ee21ef27d2
Description: Xacro (XML Macros) Xacro is an XML macro language.
 With xacro, you can construct shorter and more readable XML files by using macros that expand to larger XML expressions.
Description-md5: 8c0e70c27149552580f6b48bb7d34387
```

Anything else I might help you with?

gavanderhoorn, 2022-02-01:
Related: https://github.com/ros/xacro/pull/300#issuecomment-1025159582.

rhaschke, 2022-02-01:
Mitigated by a warning (instead of an error) in melodic-devel and noetic-devel.

caioaamaral, 2022-02-09:
I've just faced the same problem. Is there a new release coming with the fix?

rhaschke, 2022-02-10:
@caioaamaral, of course, a new release is planned. But, I simply didn't have time yet to come back here. Please note, that I am the only active maintainer of this package and I am doing all this work voluntarily in my free time. I hope to find some time over the weekend to cut a new release, but I cannot promise. If you appreciate my work, please [consider donating](https://github.com/sponsors/rhaschke).

caioaamaral, 2022-02-10:
> @caioaamaral, of course, a new release is planned.

Good to know, thanks for your effort in maintaining `xacro`

rhaschke, 2022-02-12:
Comment evaluation is opt-in now: #310. New releases are triggered:
- https://github.com/ros/rosdistro/pull/32144
- https://github.com/ros/rosdistro/pull/32145

closed 2022-02-12 by rhaschke