ros/xacro #311: xacro reads the content in comment and fail to load xacro

lianghongzhuo, 2022-02-16:
This happens in both meolodic and noetic.
When load a xacro file with comment,  it did not ignore the comment and failed to load the xacro.
To reproduce, just add a invalid comment in xacro file:
```
    <visual>
      <origin xyz="0 0 0" rpy="0 0 0" /> 
      <geometry>
        <mesh filename="package:///meshes.stl" scale="1 1 1"/>
      </geometry>
      <!-- <material name="${color}" /> -->
    </visual>
```

rhaschke, 2022-02-16:
Looks like you are using an old version. Please update to 1.14.13 (Noetic) / 1.13.17 (Melodic).
See #300 and #310 for details.

closed 2022-02-16 by rhaschke

lianghongzhuo, 2022-02-16:
my melodic is still 1.13.15. seems ros did not update yet

lianghongzhuo, 2022-02-16:
noetic is now up to date (1.14.13) and should be fine now.