ros/xacro #312: Xacro evaluating comments while running robot urdf

nksas, 2022-02-17:
I have been working on a particular robotic model/urdf, It was working fine so far. 
Recently, after "sudo apt upgrade", my xacro package got updated with the latest changes.

Soon, I realize that now when i am trying to launch any urdf model that has some big comments in it's gazebo description file, It is throwing errors because somewhere the code is trying to execute those comments.

I thought of working out with the error and i figured out that after  deleting those comments the model is able to spawn on gazebo/rviz.
So, I saw the latest commit of xacro git repository and figured that there is some minor bug in the latest code. 

I am trying to fix that but also i am submitting this issue so that if anyone else figures that our , pls submit a PR.

Following is the error that I am getting 
[image]

rhaschke, 2022-02-17:
Looks like the property `name` is not defined. You could check with very verbose mode, whether/where it gets defined:
`xacro -vvv gps.xacro`

rhaschke, 2022-02-17:
> I want to know how to explicitly turn off the comment evaluation?

Using the latest version of xacro, i.e. 1.14.13 (Noetic) / 1.13.17 (Melodic), comment evaluation is turned off by default again.

nksas, 2022-02-17:
Thank you 
I think i missed some latest upgrades on packages.
I just upgraded my xacro and it worked.

closed 2022-02-17 by nksas