ros/xacro #313: resource not found error

smileyenot983, 2022-02-17:
After recent commits i have an error with the process of converting xacro to urdf 

After i run this:
`rosrun xacro xacro robot.xacro DEBUG:=True SIMPLE_COLLISION:=True > robot_generated.urdf`
I get an error with following output:
resource not found: a1_gazebo
ROS path [0]=/opt/ros/melodic/share/ros
ROS path [1]=/home/ramil/catkin_ws/src
ROS path [2]=/opt/ros/melodic/share
when processing file: robot.xacro

rhaschke, 2022-02-17:
You are looking for a package name `a1_gazebo`. Check your files for `$(find a1_gazebo)`!
If you find this in some comments, you migth run into #311.

rhaschke, 2022-02-18:
As said before, you might run into #311. Please provide your ROS distro version as well as your xacro version for more help.

smileyenot983, 2022-02-18:
I have tried to find the place in xacro where it searches for a1_gazebo, i was commented, after deleting this comment everything works well.

ros version is:
 * /rosdistro: melodic
 * /rosversion: 1.14.12

don't know how to check xacro version

rhaschke, 2022-02-18:
I have cut a new release for Melodic recently, which is not yet synced. Consider using the debian packages from the [ROS `testing` repo](http://wiki.ros.org/TestingRepository) or build from source.

closed 2022-02-18 by rhaschke