ros/xacro #314: No attribute `main` in xacro module

sasilva1998, 2022-02-23:
Hi, I am having issues with xacro, basically can't run any launch file with a xacro file with the `robot_description` parameter. I do have another PC which is not updated and does have the `main` attribute in the xacro module.

Is there any update I am missing? Here is the output I have:

~~~~shell
sasm@sasm-XPS-17-9710:~$ roslaunch turtlebot3_gazebo turtlebot3_world.launch 
... logging to /home/sasm/.ros/log/98f060fe-94b2-11ec-896e-2c6dc1124c6f/roslaunch-sasm-XPS-17-9710-22583.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Traceback (most recent call last):
  File "/opt/ros/melodic/lib/xacro/xacro", line 33, in <module>
    xacro.main()
AttributeError: 'module' object has no attribute 'main'
RLException: Invalid <param> tag: Cannot load command parameter [robot_description]: command [['/opt/ros/melodic/lib/xacro/xacro', '--inorder', '/opt/ros/melodic/share/turtlebot3_description/urdf/turtlebot3_burger.urdf.xacro']] returned with code [1]. 

Param xml is <param command="$(find xacro)/xacro --inorder $(find turtlebot3_description)/urdf/turtlebot3_$(arg model).urdf.xacro" name="robot_description"/>
The traceback for the exception was written to the log file
~~~~

I am using:
 * /rosdistro: melodic
 * /rosversion: 1.14.12

sasilva1998, 2022-02-23:
Apparently for some reason I had a another version of xacro installed which didnt have all the files.

closed 2022-02-23 by sasilva1998

rhaschke, 2022-02-23:
I think you screwed up your ROS config. The latest available release of xacro in Melodic is 1.13.15. For Noetic it is 1.14.13.