ros/xacro #315: xacro (Melodic) parsing content within xml-comment-brackets

SeniorKullken, 2022-02-27:
#  1. Xacro Trouble Report 
 ## 1.1 Expected behaviour
 * xacro processing/interpreter **should ignore** all content in between xml-comment-brackets ``

 ## 1.2  Current behaviour
My analysis:
* Comments/Content in between xml comment brackets, `<! -` and `->`, are **also expanded/interpreted**.
    Which causes, in my example, xacro to stop and display error message `Undefined substitution argument world` . 

 ## 1.3  Background
* Recently (last month?) my old .world.xacro files stopped working => `Undefined substitution argument world` . 
* I have been using those .xacro files since the summer 2021. Without any changes. 
  But  now in jan/deb-2022 they have stop working.

## 1.4 My work environment
* **Melodic** with the latest updates (Today is: 27 feb, 2022).
  _Note! - It looks like it works with Noetic..._
* **Ubuntu 18.04**(64-bit)

# 2. Examples
See example below "How to repeat the issue", in Melodic, and "Temporary workaround" I made.
(I have cut down the attached examples and ,almost ,only kept the xml comments that causing the error)

## 2.1 Example: How to repeat the issue

Output from: `$ cat TroubleReport_error.xacro`  
Attached: [TroubleReport_error.xacro](https://github.com/ros/xacro/files/8148691/TroubleReport_error.xacro.txt)
```xml
<?xml version="1.0" ?> 
<sdf version="1.5"> 

  <!-- 
    ".xacro" => SDF/URDF-macro file that need to be parsed before loading into Gazebo. 

    Example - Via shell/command line: 
      $ rosrun xacro xacro src/Pet-Mk-IV/pet_mk_iv_simulation/worlds/pet_play_yard.world.xacro >  src/Pet-Mk-IV/pet_mk_iv_simulation/worlds/pet_play_yard.world  

    Example - Via launch file: 
      <node name="xacro" pkg="xacro" type="xacro" args="$(find pet_mk_iv_simulation)/worlds/$(arg world).xacro -o $(find pet_mk_iv_simulation)/worlds/$(arg world)" /> 

    http://wiki.ros.org/xacro  
  --> 

  <world name="default"> 
    <!-- REMOVED ALL THE CONTENT IN DEBUGGING PURPOSE... ONLY LEAVING A EMPTY WORLD --> 

    <!-- A global light source --> 
    <include> 
      <uri>model://sun</uri> 
    </include> 

    <!-- A ground plane --> 
    <include> 
      <uri>model://ground_plane</uri> 
    </include> 
  </world> 
</sdf>
```
Output from: `$ rosrun xacro xacro TroubleReport_error.xacro `  
```shell
Undefined substitution argument world 
when processing file: TroubleReport_error.xacro 
```
## 2.2 Example: A temporary workaround
As a workaround, I  replaced `$` with `[** dollars **]` in between the xml-comment-brackets ``.

Output from: `$ cat TroubleReport_workaraound.xacro`  
Attached: [TroubleReport_workaraound.xacro](https://github.com/ros/xacro/files/8148692/TroubleReport_workaraound.xacro.txt)
```xml
<?xml version="1.0" ?> 
<sdf version="1.5"> 

  <!-- 
    ".xacro" => SDF/URDF-macro file that need to be parsed before loading into Gazebo. 

    Example - Via shell/command line: 
      $ rosrun xacro xacro src/Pet-Mk-IV/pet_mk_iv_simulation/worlds/pet_play_yard.world.xacro >  src/Pet-Mk-IV/pet_mk_iv_simulation/worlds/pet_play_yard.world  

    Example - Via launch file: 
      <node name="xacro" pkg="xacro" type="xacro" args="$(find pet_mk_iv_simulation)/worlds/[**dollar**](arg world).xacro -o $(find pet_mk_iv_simulation)/worlds/[**dollar**](arg world)" /> 

    http://wiki.ros.org/xacro  
  --> 

Pay attention to how `$(find pet_mk_iv_simulation)` is handled  => Expanded to ` /home/sk/ros_ws/src/pet_mk_iv_simulation` 
  <world name="default"> 
    <!-- REMOVED ALL THE CONTENT IN DEBUGGING PURPOSE... ONLY LEAVING A EMPTY WORLD --> 

    <!-- A global light source --> 
    <include> 
      <uri>model://sun</uri> 
    </include> 

    <!-- A ground plane --> 
    <include> 
      <uri>model://ground_plane</uri> 
    </include> 
  </world> 
</sdf>
```

Output from: `$ rosrun xacro xacro TroubleReport_workaraound.xacro`  
Attached: [TroubleReport_workaraound.world](https://github.com/ros/xacro/files/8148696/TroubleReport_workaraound.world.txt)
```xml
<?xml version="1.0" encoding="utf-8"?>
<!-- =================================================================================== -->
<!-- |    This document was autogenerated by xacro from TroubleReport_workaraound.xacro.txt | -->
<!-- |    EDITING THIS FILE BY HAND IS NOT RECOMMENDED                                 | -->
<!-- =================================================================================== -->
<sdf version="1.5">
  <!-- 
    ".xacro" => SDF/URDF-macro file that need to be parsed before loading into Gazebo. 

    Example - Via shell/command line: 
      $ rosrun xacro xacro src/Pet-Mk-IV/pet_mk_iv_simulation/worlds/pet_play_yard.world.xacro >  src/Pet-Mk-IV/pet_mk_iv_simulation/worlds/pet_play_yard.world  

    Example - Via launch file: 
      <node name="xacro" pkg="xacro" type="xacro" args="/home/sk/ros_ws/src/pet_mk_iv_simulation/worlds/[**dollar**](arg world).xacro -o /home/sk/ros_ws/src/pet_mk_iv_simulation/worlds/[**dollar**](arg world)" /> 

    http://wiki.ros.org/xacro  
  -->
  <world name="default">
    <!-- REMOVED ALL THE CONTENT IN DEBUGGING PURPOSE... ONLY LEAVING A EMPTY WORLD -->
    <!-- A global light source -->
    <include>
      <uri>model://sun</uri>
    </include>
    <!-- A ground plane -->
    <include>
      <uri>model://ground_plane</uri>
    </include>
  </world>
</sdf>
```
## 2.3 Attached files
Attached: [TroubleReport_error.xacro](https://github.com/ros/xacro/files/8148691/TroubleReport_error.xacro.txt)
Attached: [TroubleReport_workaraound.xacro](https://github.com/ros/xacro/files/8148692/TroubleReport_workaraound.xacro.txt)
Attached: [TroubleReport_workaraound.world](https://github.com/ros/xacro/files/8148696/TroubleReport_workaraound.world.txt)

rhaschke, 2022-02-27:
See https://github.com/ros/xacro/issues/309#issuecomment-1037236018 for a fix. Unfortunately, Melodic hasn't seen a release sync for a while. Consider using the debian packages from the [ROS `testing` repo](http://wiki.ros.org/TestingRepository) or build xacro from source.

rhaschke, 2022-04-21:
Closing as Melodic as well as Noetic distros have synced the latest fixes.

closed 2022-04-21 by rhaschke