ros/xacro #317: load_yaml dot accessor breaks down in loops

Doomerdinger, 2022-06-09:
I've notice that if I call (from python code, not within a xacro file) `load_yaml` the dot accessor does not work in for loops.

Given this yaml file:
```yaml
---
list_ex:
  - name: foo
    val:
      x: 1
  - name: bar
    val:
      x: 2
```

The following will not work correctly:
```python
b = 0
loaded_yaml = load_yaml(file)
for o in loaded_yaml.list_ex:
    b = b + o.val.x
```
It will complain that `'dict' has no value 'val'`

However, the following will work fine
```python
b = 0
loaded_yaml = load_yaml(file)
for x in range(0, len(loaded_yaml.list_ex)):
    o = loaded_yaml[x]
    b = b + o.val.x
```

Calling this code directly is not the intended use case, so this may not be worth addressing.

rhaschke, 2022-06-09:
`load_yaml` uses two wrapper classes, [`YamlListWrapper`](https://github.com/ros/xacro/blob/a8738a4c632fc767c05b9a3bf55209b71ea65f91/src/xacro/__init__.py#L86) and [`YamlDictWrapper`](YamlDictWrapper) to allow dotted access to elements. To this end, the actually retrieved element is wrapped within one of these classes depending on its type.
For now, element access is correctly implemented. To enable your use case, you would need to implement a wrapper for the iterator returned via `__iter__()` as well. Otherwise, the info that a wrapper is desired is lost. I will have a look...

Doomerdinger, 2022-06-09:
Your change seems to make this work properly

rhaschke, 2022-06-09:
Fixed via #318

PR #318 (2022-06-09): Dotted YAML access from list iterator
Fixes #317
closed 2022-06-09 by rhaschke