ros/xacro #325: xacro_add_files fails if another file in the current building package is included in the file

furushchev, 2023-07-04:
The cmake function `xacro_add_files` fails when the target `xacro` file has include tags which has the path to files in the current package.
It looks this is because we cannot find the current building package using `ament_index` system.
Is there any fix or workaround?

Here is how to reproduce the issue:

Given a `test_pkg` in a workspace:

- src/test_pkg/CMakeLists.txt

```cmake
cmake_minimum_required(VERSION 3.8)
project(test_pkg)

find_package(ament_cmake REQUIRED)
find_package(xacro REQUIRED)

xacro_add_files(foo.urdf.xacro TARGET foo)

ament_package()
```

- src/test_pkg/foo.urdf.xacro

```xml
<?xml version="1.0" encoding="utf-8"?>
<robot name="robot" xmlns:xacro="http://www.ros.org/wiki/xacro">
  <xacro:include filename="$(find test_pkg)/bar.urdf.xacro"/>
</robot>
```

- src/test_pkg/bar.urdf.xacro

```xml
<?xml version="1.0" encoding="utf-8"?>
<robot xmlns:xacro="http://www.ros.org/wiki/xacro">
</robot>
```

Run `colcon build` fails on error:

```
$ source /opt/ros/humble/setup.bash
$ colcon build --symlink-install
Starting >>> test_pkg
--- stderr: test_pkg                              
CMake Warning at /opt/ros/humble/share/xacro/cmake/xacro-extras.cmake:74 (message):
  failed to determine deps for: foo.urdf.xacro

  error: package not found: "package 'test_pkg' not found, searching:
  ['/opt/ros/humble']"

  when processing file: foo.urdf.xacro

Call Stack (most recent call first):
  /opt/ros/humble/share/xacro/cmake/xacro-extras.cmake:133 (xacro_add_xacro_file)
  CMakeLists.txt:12 (xacro_add_files)

package not found: "package 'test_pkg' not found, searching: ['/opt/ros/humble']"
when processing file: foo.urdf.xacro
gmake[2]: *** [CMakeFiles/test_pkg_foo.dir/build.make:73: foo.urdf] Error 2
gmake[1]: *** [CMakeFiles/Makefile2:137: CMakeFiles/test_pkg_foo.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< test_pkg [0.47s, exited with code 2]

Summary: 0 packages finished [0.57s]
  1 package failed: test_pkg
  1 package had stderr output: test_pkg
```

rhaschke, 2023-07-04:
I don't have a clean solution to this problem. See #326 for an intermediate workaround.

furushchev, 2023-07-05:
Thank you for addressing this issue! I hope there should be a way that the current package path is resolved at build time.

rhaschke, 2023-07-05:
> The downside of #326 is if any custom destination is set in `install` function in `CMakeLists.txt`, it will not work.

@furushchev, can you please elaborate on this statement? I don't yet see a limitation.

furushchev, 2023-07-05:
@rhaschke 
Given that there is `urdf` directory in the project, if the `install` function is set as followings:

```
install(DIRECTORY urdf DESTINATION share/${PROJECT_NAME}/resources)
```

The file can only be looked up as `$(find pkg)/resources/urdf/foo.urdf.xacro`, which is different from the location in the source directory `$(find pkg)/urdf/foo.urdf.xacro`.

However, it is really edge case, so I think we can ignore it anyway.

rhaschke, 2023-07-05:
I see. But the same issue arises in ROS1: build-space and install-space paths have to be the same.

furushchev, 2023-07-05:
> But the same issue arises in ROS1: build-space and install-space paths have to be the same.

Thank you for the answer. Yes, I just now realized the same thing, sorry for confusion.

rhaschke, 2023-12-23:
Closed via #326.

PR #326 (2023-10-31): HACK: Allow resolution of current project's path at build time
The cmake function `xacro_add_files` runs `xacro` at build time. To resolve `$(find current_package)` within xacro files, `xacro` relies on a properly configured ament package index. However, this index is only created after building, during installation.
By manually prepending `AMENT_PREFIX_PATH` with the current project's `ament_cmake_index` in the `build` folder _and_ symlinking the source folder into the corresponding `share` folder, I was able to work around this shortcoming. However, a fix in https://github.com/ament/ament_cm [... cut]
closed 2023-12-23 by rhaschke