ros/xacro #328: PyPi Xacro is outdated

hubernikus, 2023-12-22:
The PyPi package of xacro is of version 1.13.3 - but on this github there have been many new releases since.

Is there a plan to update PyPi (for usage without the complete ros-library), or how should xacro be used?

hubernikus, 2023-12-22:
After digging a bit further, it seems the Pypi version of xacro has been independently developed from this repository. Will there ever be a version of xacro which does not require ROS for the installation?

gavanderhoorn, 2023-12-22:
This has been discussed a couple times in the past. See #197 fi.

I'm not a maintainer here. I just thought I'd save @rhaschke the trouble of having to look up that issue again.

gavanderhoorn, 2023-12-22:
It's been a while since #197 was posted though. So perhaps something has changed.

Let's see what @rhaschke says.

hubernikus, 2023-12-22:
For the installation of the package in a ros agnostic way, it seems to lack a 'setup.py' or 'pyproject.toml' file (as well as requirements, which is only 'pyyaml' I believe). Is there any intention to add these? As it would allow direct installation from the github repo using pip or others.

rhaschke, 2023-12-23:
As Gijs wrote, the PyPi package is not an official one. 
xacro does have a `setup.py` and installing directly from github works for me: 
`pip install git+https://github.com/ros/xacro.git@noetic-devel`. 
dependency declarations are missing therein. I'm also not sure whether these would conflict with Ubuntu package dependencies declared via ROS in package.xml. I'm looking forward to your PR adding any missing stuff ;-)

hubernikus, 2023-12-24:
True,  the 'noetic-devel' branch has a `setup.py` file. However, there is a dependency on ros1 / catking, and I get a dependency error,  ` ModuleNotFoundError: No module named 'catkin_pkg'`.

The `ros2` branch does not have a setup file anymore. I'll make a pull request to add it.

closed 2024-02-03 by rhaschke