ros/xacro #335: Running roslaunch franka_gazebo panda.launch gives name 'xacro' not defined error

giorgiapazzi, 2024-02-16:
I'm running the example reported in franka_ros documentation [](https://frankaemika.github.io/docs/franka_ros.html#franka-gazebo).
When I try to run:
`roslaunch franka_gazebo panda.launch x:=-0.5 \
    world:=$(rospack find franka_gazebo)/world/stone.sdf \
    controller:=cartesian_impedance_example_controller \
    rviz:=true`

I get the following error:
`name 'xacro' is not defined 
when evaluating expression 'xacro.load_yaml('$(find franka_description)/robots/panda/joint_limits.yaml')'
`
I get the same error launching:
`xacro $(rospack find franka_description)/robots/panda/panda.urdf.xacro gazebo:=true`

I'm using franka_ros for Ros Noetic.

rhaschke, 2024-02-16:
This works for me as expected. Which version of xacro do you use? Maybe you actually use an old version?

giorgiapazzi, 2024-02-16:
I'm using the version 1.14.6 for xacro

giorgiapazzi, 2024-02-16:
I have installed libfranka using the commando sudo apt install libfranka.
Instead, for the franka_ros package, I have installed it from the source code cloning the git repository branch noetic-devel.

giorgiapazzi, 2024-02-16:
which xacro command returns:
`/opt/ros/noetic/bin/xacro`

I have tried also rospack find xacro and the result is:
`/opt/ros/noetic/share/xacro`

rhaschke, 2024-02-16:
This is weird. Could you try reinstalling xacro? What happens if you use xacro from source as well?
Do you use a virtual or conda environment?

giorgiapazzi, 2024-02-19:
I solved the problem by cloning the xacro repository directly into my workspace and remake.

rhaschke, 2024-02-19:
If that works, your installation of xacro is somehow screwed. Can we close this issue?

giorgiapazzi, 2024-02-19:
Yes we can close it

closed 2024-02-19 by rhaschke