ros/xacro #337: Unable to resolve namespace within property block and not able to pass a block by a macro to

naimen, 2024-02-28:
Hi I've been playing around with insert_block and property blocks, and discovered that it can't evaluate the following example, where it throws error of 
``` 
unknown macro name: xacro:CH.bar
when instantiating macro: insert (child.xml)
in file: parent.xml

``` 

parent.xml
```  
    <xacro:include filename="child.xml" ns="CH"/>

    <xacro:property name="A_block" scope="global" lazy_eval="false">
        <xacro:CH.barA="-1" B="2"/>
    </xacro:property>

    <xacro:CH.insert/>

    <xacro:CH.insert_2>
        <foo>
            <xacro:CH.barA="-1" B="2"/>
        </foo>
    </xacro:CH.insert_2>
``` 

child.xml
```  
    <xacro:macro name="bar" params="A B">
        <bar data="${A} 0 1 ${B}"/>
        <bar data="${B} 1 0 ${A}"/>
    </xacro:macro>

    <xacro:macro name="insert">
        <foo>
            <xacro:insert_block name="A_block"/>
        </foo>
    </xacro:macro>

    <xacro:macro name="insert_2" params="**foo">
        <foo>
            <xacro:insert_block name="foo"/>
        </foo>
    </xacro:macro>   
```

Any idea of how to handle this?

rhaschke, 2024-02-28:
You must not use the namespace `CH` when referring to the macro `bar` within `child.xml`, because the macro is already executing within that namespace:
```xml
<parent xmlns:xacro="http://ros.org/wiki/xacro">
	<xacro:include filename="child.xml" ns="CH"/>

	<xacro:property name="A_block" scope="global" lazy_eval="false">
		<xacro:bar A="-1" B="2"/>
	</xacro:property>

	<xacro:CH.insert/>

	<xacro:CH.insert_2>
		<foo>
			<xacro:bar A="-1" B="2"/>
		</foo>
	</xacro:CH.insert_2>
</parent>
```

naimen, 2024-02-29:
```> You must not use the namespace `CH` when referring to the macro `bar` within `child.xml`, because the macro is already executing within that namespace:
> ```
> <parent xmlns:xacro="http://ros.org/wiki/xacro">
> 	<xacro:include filename="child.xml" ns="CH"/>
>
> 	<xacro:property name="A_block" scope="global" lazy_eval="false">
> 		<xacro:bar A="-1" B="2"/> [...]

Thanks for the reply, so when I'm calling the macro I will be in scope of the called macro(or file)? Because if i move the macro `bar` to the `parent.xml`, then xacro fails with 

```
unknown macro name: xacro:bar
when instantiating macro: insert (child.xml)
in file: parent.xml
```

rhaschke, 2024-02-29:
Sure. The block is interpreted within the context of the macro. If the macro is in the namespace, you don't need to add the namespace again. If the macro is moved out from the namespace, you need to specify the namespace explicitly.

naimen, 2024-03-01:
> Sure. The block is interpreted within the context of the macro. If the macro is in the namespace, you don't need to add the namespace again. If the macro is moved out from the namespace, you need to specify the namespace explicitly.

If my macro `bar` is inside `parent`, then when i'm executing the macro `child.insert` where it points to a property block defined in `parent` that calls `bar` without lazy_eval, I will be in the context of `insert` within `child` and performes the evaulation of property block where i calls `bar` from that context? 

So there's no way for me to pass block created by a macro from a context into a another macro in another context?

rhaschke, 2024-03-01:
> So there's no way for me to pass block created by a macro from a context into a another macro in another context?

No, I'm afraid that's not possible.

naimen, 2024-03-01:
> Correctly pass the namespace.

You can pass namespace? :open_mouth: 
Do you have an example of pass `ns1` to a macro in `ns2` without explicity includ that `ns1` inside `ns2`?

rhaschke, 2024-03-01:
As I wrote, you cannot cross-reference namespaces. Namespaces were introduced to distinguish two child-namespaces from within a common parent. They are not intended to access ns1 from within ns2.

closed 2024-03-10 by rhaschke