ros/xacro #338: $(find) no longer resolved after other args

adamheins, 2024-02-29:
On the `noetic-devel` branch, resolving substitution arguments was outsourced to roslaunch. This resolves `$(find)` after other arguments like `$(arg)` (see [here](https://github.com/ros/ros_comm/blob/noetic-devel/tools/roslaunch/src/roslaunch/substitution_args.py#L380-L392)), which means I do things like pass `$(find)` into an argument. For example:
```xml
<xacro:arg name="robotfile" default="default.urdf.xacro"/>
<xacro:include filename="$(arg robotfile)"/>
```
and then I could pass in something like `robotfile:=$(find mypkg)/urdf/robot.urdf.xacro`, and expect the included filename to ultimately be resolved properly through the `$(arg)` and `$(find)`.

On the `ros2` branch, this logic has been changed so that the above no longer works for me. I'm wondering if this change was intended, or if I'm just doing something wrong here.

rhaschke, 2024-03-01:
Fixed via #339. Do you need a release? Into which ROS2 distros?

adamheins, 2024-03-01:
Thanks, that does it! I don't actually need a release; I came across this while working on [xacrodoc](https://github.com/adamheins/xacrodoc), which is a small tool that modifies xacro to (1) have more support for programmatically compiling files, and (2) makes the package-lookup mechanism fully functional even without ROS.

PR #339 (2024-03-01): Resolve $(find ...) as a result of a substitution argument
Fixes #338
The implementation of substitutions_args introduced in [7310f9a](https://github.com/ros/xacro/pull/213/commits/7310f9a1fb4fa9db11b60c6dced6b255c265229e) as part of #213 was missing the second call to resolve `$(find ...)` arguments.
closed 2024-03-01 by adamheins