ros/xacro #342: xacro parser unable to parse xml attributes containing a colon

davidorchansky, 2024-05-10:
### Description

It seems that when attempting to parse a xacro file that contains an xml tag with an attribute that requires a colon, the parsing of the file fails with the following error:

`ERROR: unbound prefix: line 29, column 4`

### Background

I'm trying to create a xacro file that contains a custom sensor similarly to the one defined here:
[https://github.com/gazebosim/gz-sim/blob/gz-sim8/test/worlds/underwater_currents.sdf](https://github.com/gazebosim/gz-sim/blob/gz-sim8/test/worlds/underwater_currents.sdf)

The relevant section is below:
```
<sensor element_id="base_link" action="add"
            name="teledyne_pathfinder_dvl"
            type="custom" gz:type="dvl">
...
</sensor>
```
It seems that the xacro parser does not like the attribute **gz:type** as it contains a colon. 

### Expected Behavior
The xacro file containing attributes with a colon are parsed with no problem.

### Environment
  xacro package version: 2.0.8

### Steps to reproduce

Create test.xacro containing:

```
<?xml version="1.0"?>
<robot name="test" xmlns:xacro="http://www.ros.org/wiki/xacro" >
  <gazebo reference="base_link">
    <!-- Based on https://github.com/gazebosim/gz-sim/blob/gz-sim8/test/worlds/underwater_currents.sdf -->
    <sensor
        element_id="base_link" action="add"
        name="teledyne_pathfinder_dvl"
        type="custom" gz:type="dvl">
      <pose degrees="true">-0.60 0 -0.16 0 0 180</pose>
      <always_on>1</always_on>
      <update_rate>1</update_rate>
      <topic>/dvl/velocity</topic>
      <gz:dvl>
        <type>phased_array</type>
        <arrangement degrees="true">
          <beam id="1">
            <aperture>2</aperture>
            <rotation>45</rotation>
            <tilt>30</tilt>
          </beam>
          <beam>
            <aperture>2</aperture>
            <rotation>135</rotation>
            <tilt>30</tilt>
          </beam>
          <beam>
            <aperture>2</aperture>
            <rotation>-45</rotation>
            <tilt>30</tilt>
          </beam>
          <beam>
            <aperture>2</aperture>
            <rotation>-135</rotation>
            <tilt>30</tilt>
          </beam>
        </arrangement>
        <tracking>
            <bottom_mode>
              <when>best</when>
              <noise type="gaussian">
                <!-- +/- 0.4 cm/s precision at 10 m/s within 2 stddevs -->
                <stddev>0.002</stddev>
              </noise>
              <visualize>false</visualize>
            </bottom_mode>
            <water_mass_mode>
              <when>best</when>
              <water_velocity>
                <x>eastward_sea_water_velocity_meter_per_sec</x>
                <y>northward_sea_water_velocity_meter_per_sec</y>
              </water_velocity>
              <boundaries>
                <near>20.</near>
                <far>60.</far>
              </boundaries>
              <bins>10</bins>
              <noise type="gaussian">
                <!-- +/- 0.7 cm/s precision at 10 m/s within 2 stddevs -->
                <stddev>0.0035</stddev>
              </noise>
              <visualize>false</visualize>
            </water_mass_mode>
          </tracking>
          <!-- Roughly 1 m resolution at a 100m -->
          <resolution>0.01</resolution>
          <maximum_range>100.</maximum_range>
          <minimum_range>0.1</minimum_range>
          <!-- ENU to SFM -->
          <reference_frame>0 0 0 0 0 -1.570796</reference_frame>
      </gz:dvl>
    </sensor> 
  </gazebo>
</robot>
```

Create a launch file called test.launch.py with the following content:

```
from launch import LaunchDescription
from launch.actions import GroupAction
from launch.actions import OpaqueFunction

import os
import xacro

def launch_setup(context, *args, **kwargs): 
    xacro_file = <path to test.xacro>
    doc = xacro.parse(open(xacro_file))
    xacro.process_doc(doc, mappings=mappings)

    return [GroupAction([])]

def generate_launch_description():
    return LaunchDescription([
        OpaqueFunction(function = launch_setup)
    ])
```

Then launch the launch file.

rhaschke, 2024-05-12:
You simply missed to declare the namespace for prefix `gz`: `xmlns:gz="whatever.this.namespace.refers.to"`.
The prefix before a colon refers to a namespace declaration, which needs to be available in the document...
The error message hints you at that problem.

closed 2024-05-12 by rhaschke

davidorchansky, 2024-05-12:
Oh, I missed that! Thank you @rhaschke.