ros/xacro #345: Xacro file argument introspection

reinzor, 2024-06-11:
We would like to add a feature that allows a user to introspect xacro files. This way, we can query what arguments are available and prompt these to the user. We could also add additional validation to the defined arguments and some documentation. What are your thoughts about this? An example:

```xml
<xacro:arg name="argument1" default="value1" doc="Argument 1 doc" choices="value1, value2, value3" />
```

```bash
xacro --args test.xacro
```

```
arguments:
  - name: argument 1
  - default: value1
  - choices: [value1, value2, value3]
  - doc: Argument 1 doc
```

rhaschke, 2024-06-11:
This is generally a good idea. However, as of now, arguments don't have a `choices` attribute yet and I'm not sure it makes sense to introduce it. Often, arguments can take arbitrary values, such that a choices restriction doesn't make sense. Another restriction option is a range of numbers, which is difficult to express as a discrete list of choices. In both cases, the allowed parameters could be easily specified within the doc string. Of course, this makes generic validation impossible, but that's impossible anyway given the multitude of possible constraints. Note that validation is possible with corresponding code already.
Regarding the output formatting, I'd prefer a more compact format, like:
```
arguments:
  - argument1=default: doc
  - ...
```

reinzor, 2024-06-11:
Thanks; I agree with your comment on the choices attribute. I will try to prepare a PR that adds the argument introspection and the `doc` attribute.

reinzor, 2024-06-11:
Another though that comes to mind, couldn't we add a `validation` attribute that accepts an expression as defined here: https://github.com/ros/xacro/wiki#expression-evaluation- . would this work? For example:

```
<arg name="arg1" validation="${xacro.arg('arg1') in ['value1', 'value2']}" />
```

rhaschke, 2024-06-11:
This would be more generic at least.