ros/xacro #347: Importing Same Robot Multiple Times

hubernikus, 2024-07-19:
I want to use robot to parse an urdf file which contains the same robot (but at different positions).

I found on the wiki the option of 
<xacro:include filename="other.xacro" ns="other"/>

However, I would need some kind of name-prefix to be added to the links/joints. 

Additionally, I also want to use the same robot model for single robot setups (i.e. without name-prefix). Hence, ideally, I would like to have a robot model. 

Is there an easy way to do this?

Would it be useful to add a 'nameprefix' argument to the 'xacro:include' which automatically updates the to be included files?

rhaschke, 2024-07-19:
You need to generate your robot description via a macro, which takes the name_prefix argument to modify link and joint names. Examples for this can be found on the web, e.g. [franka_description](https://github.com/frankaemika/franka_description/blob/main/robots/common/franka_arm.xacro). The `xacro:include` tag is not suitable for this: its namespace argument just introduces a namespace for xacro.

hubernikus, 2024-07-19:
Ok. Thanks a lot. And there is no way of checking if a variable is defined using the conditional-block, is there?

rhaschke, 2024-07-19:
> And there is no way of checking if a variable is defined using the conditional-block, is there?

I don't think so.

hubernikus, 2024-07-19:
Indeed, https://github.com/ros/xacro/blob/noetic-devel/src/xacro/__init__.py#L251 removes all access to build ins. 
Which makes sense for security.

However, one could additionally expose the local-namespace, i.e., just giving access to the local variable dictionary, and without exposing any other built ins / additional functionality.
```python
globals.update(__builtins__= {"vars": vars})  # disable default builtins
```

This would then enable variable checking in the xacro, and hence enable the defining of 'default' parameters:
```xacro
<xacro:if value='${"foo" not in vars()}'>
    <xacro:property name="foo" value="default_value" />
<xacro:if value='${"foo" not in vars()}'>
```

hubernikus, 2024-07-19:
Probably better to add it to the list directly:
https://github.com/ros/xacro/blob/noetic-devel/src/xacro/__init__.py#L212

PR #348 (2024-07-19): Add function python.vars()
Fixes #347
closed 2024-07-19 by rhaschke