ros/xacro #354: is it possible to change the name in robot tag using a parameter?

xibeisiber, 2025-01-16:
hi, in a urdf.xacro, the typical routine is ：
```
<?xml version='1.0' encoding='utf-8'?>
<robot xmlns:xacro="http://www.ros.org/wiki/xacro" name="robname">
   <xacro: arg name="id" default="0/>
   ---blabla---
</robot>
```
It is convenient to change the value of 'id' in launch file or using `xacro a.urdf.xacro id:=1 -o a.urdf`.

Is it possible to change the name in robot tag? 
it seems it can't be declared as a xacro arg or property since `xmlns:xacro` is defined in the same tag.

rhaschke, 2025-01-16:
Yes, it's possible to use `$(arg ...)` syntax in the top-level tag:
`<robot xmlns:xacro="http://www.ros.org/wiki/xacro" name="$(arg name)"/>`
However, you can't declare the argument (and provide a default value). 
Instead, you _need_ to pass the argument via the command line:
`xacro top.xacro name:="foo"`

xibeisiber, 2025-01-17:
ok. Thanks!

closed 2025-01-17 by xibeisiber