ros/xacro #364: Error: No link elements found in urdf file

lxq306, 2025-05-16:
I'd use a simple demo to launch a xacro file, and got the Error:

[image]

$ROS_DISTRO = "jazzy"

Any suggestion for that? Thanks a lot!

rhaschke, 2025-05-16:
Please avoid posting images if you actually want to post text. Images are not searchable!
Your sample .urdf.xacro doesn't call a macro. Thus, the generated .urdf is essentially empty.

lxq306, 2025-05-19:
> Please avoid posting images if you actually want to post text. Images are not searchable! Your sample .urdf.xacro doesn't call a macro. Thus, the generated .urdf is essentially empty.

Sorry about the images and thanks a lot for your reply! I repost the code here: 
<?xml version="1.0"?>
<robot xmlns:xacro="http://www.ros.org/wiki/xacro" name="first_robot">
    
    <xacro:macro name="base_link" params="length radius" >
        <link name="base_link">
            <visual>
                <origin xyz="0 0 0" rpy="0 0 0" />
                <geometry>
                    <cylinder length="${length}" radius="${radius}" />
                </geometry>
                <material name="white">
                    <color rgba="1.0 1.0 1.0 0.5" />
                </material>
            </visual>
        </link>
    </xacro:macro>
	
    **<xacro:base_link length="0.12" radius="0.1" />**  
</robot>

Isn't "<xacro:base_link length="0.12" radius="0.1">" a macro call? Really appreciate !!!

rhaschke, 2025-05-19:
You are right. This .xacro file translates to:
```
<robot name="first_robot">
  <link name="base_link">
    <visual>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry>
        <cylinder length="0.12" radius="0.1"/>
      </geometry>
      <material name="white">
        <color rgba="1.0 1.0 1.0 0.5"/>
      </material>
    </visual>
  </link>
  <!--Isn't this a macro call?-->
</robot>
```