ros/xacro #365: Error: No link elements found in urdf file

lxq306, 2025-05-19:
I'd use a simple demo to launch a xacro file, and got the Error: No link elements found in urdf file:

```
<?xml version="1.0"?>
<robot xmlns:xacro="http://www.ros.org/wiki/xacro" name="first_robot">
    <!--base_link-->
    <xacro:macro name="base_link" params="length radius" >
        <link name="base_link">
            <visual>
                <origin xyz="0 0 0" rpy="0 0 0" />
                <geometry>
                    <cylinder length="${length}" radius="${radius}" />
                </geometry>
                <material name="white">
                    <color rgba="1.0 1.0 1.0 0.5" />
                </material>
            </visual>
        </link>
    </xacro:macro>
	
    <xacro:base_link length="0.12" radius="0.1" />  <!--macro call-->
</robot>
```
I do call the macro "base_link", and I got the “No link elements found in urdf file”error:

```
1747625838.2326946 [INFO] [launch]: Default logging verbosity is set to INFO
1747625838.3411953 [INFO] [joint_state_publisher-1]: process started with pid [6664]
1747625838.3415649 [INFO] [robot_state_publisher-2]: process started with pid [6665]
1747625838.3417678 [INFO] [rviz2-3]: process started with pid [6666]
**1747625838.4327872 [robot_state_publisher-2] Error:   No link elements found in urdf file**
1747625838.4330630 [robot_state_publisher-2]          at line 207 in ./urdf_parser/src/model.cpp
1747625838.4332459 [robot_state_publisher-2] Failed to parse robot description using: urdf_xml_parser/URDFXMLParser
1747625838.4333820 [robot_state_publisher-2] terminate called after throwing an instance of 'std::runtime_error'
1747625838.4334736 [robot_state_publisher-2]   what():  Unable to initialize urdf::model from robot description
1747625838.8576713 [rviz2-3] [INFO] [1747625838.857216070] [rviz2]: Stereo is NOT SUPPORTED
1747625838.8583798 [rviz2-3] [INFO] [1747625838.857362864] [rviz2]: OpenGl version: 3.1 (GLSL 1.4)
1747625838.8844106 [rviz2-3] [INFO] [1747625838.884025537] [rviz2]: Stereo is NOT SUPPORTED
1747625839.2293777 [joint_state_publisher-1] [INFO] [1747625839.228841387] [joint_state_publisher]: Waiting for robot_description to be published on the robot_description topic...
1747625839.7376368 [ERROR] [robot_state_publisher-2]: process has died [pid 6665, exit code -6, cmd '/opt/ros/jazzy/lib/robot_state_publisher/robot_state_publisher --ros-args --params-file /tmp/launch_params_jcflddwo'].
1747625841.2495100 [INFO] [rviz2-3]: process has finished cleanly [pid 6666]
1747626209.0295994 [WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
1747626209.2901893 [INFO] [joint_state_publisher-1]: process has finished cleanly [pid 6664]
```
$ROS_DISTRO = "jazzy"
Any suggestion? Really appreciate for your help!!!

rhaschke, 2025-05-19:
Looks like the urdf parser is broken in your distro. For me (on Humble), the following works without issues:
`ros2 run robot_state_publisher robot_state_publisher --ros-args -p robot_description:="$(xacro /tmp/test.xacro)"`

I suggest to file a bug report with https://github.com/ros2/urdf.

closed 2025-05-19 by rhaschke

lxq306, 2025-05-20:
I‘ll try, thanks a lot :-)