ros/xacro #366: Xacro namespace pollution

AntoSave, 2025-06-04:
I believe I’ve encountered a namespace pollution issue when using `<xacro:include>` with the `ns` attribute. I am using xacro version **2.0.13**.

### Description
I have set up a minimal example to reproduce the behavior I have observed, which consists of the following xacro files:

#### `A.xacro`
```xml
<?xml version="1.0" encoding="utf-8"?>
<a xmlns:xacro="http://www.ros.org/wiki/xacro">
  <xacro:property name="test_property" value="test_value_A"/>
</a>
```

#### `B.xacro`
```xml
<?xml version="1.0" encoding="utf-8"?>
<b xmlns:xacro="http://www.ros.org/wiki/xacro">
  <xacro:property name="test_property" value="test_value_B"/>
  ${xacro.message('test_property: ' + test_property)}
  <xacro:include filename="./A.xacro" ns="A"/>
  ${xacro.message('A.test_property: ' + A.test_property)}
  ${xacro.message('test_property: ' + test_property)}
</b>
```

### Expected Output

When using the `ns` attribute of the `xacro:include` tag, I expect the local namespace of file `B` to be unchanged. Therefore I was expecting the following output:

```
test_property: test_value_B
A.test_property: test_value_A
test_property: test_value_A
```

### Actual Output

```
test_property: test_value_B
A.test_property: test_value_A
test_property: test_value_B
```

It looks like file `B`'s local namespace is polluted by the properties define within file `A`.

I find this behavior to be counterintuitive and may suggest a bug or an ambiguity in how namespaces and property resolution are handled in xacro.

Please let me know if this is the intended behavior or if it's something that should be fixed.

Thanks!

AntoSave, 2025-06-04:
Sorry, I must have messed up while trying to simplify the example I provided. The actual example was the following:

#### `A.xacro`
 ```xml
<?xml version="1.0" encoding="utf-8"?>
<a xmlns:xacro="http://www.ros.org/wiki/xacro">
  <xacro:property name="test_property" value="test_value_A"/>

  <xacro:macro name="test_macro" params="test_param">
    ${xacro.message('test_param: ' + test_param)}
  </xacro:macro>
</a>
```

#### `B.xacro`
```xml
<?xml version="1.0" encoding="utf-8"?>
<b xmlns:xacro="http://www.ros.org/wiki/xacro">
  <xacro:property name="test_property" value="test_value_B"/>
  ${xacro.message('test_property: ' + test_property)}
  <xacro:include filename="./A.xacro" ns="A"/>
  ${xacro.message('A.test_property: ' + A.test_property)}
  ${xacro.message('test_property: ' + test_property)}
  <xacro:A.test_macro test_param="${test_property}"/>
</b>
```

### Expected Output
```
test_property: test_value_B
A.test_property: test_value_A
test_property: test_value_B
test_param: test_value_B
```

### Actual Output
```
test_property: test_value_B
A.test_property: test_value_A
test_property: test_value_B
test_param: test_value_A
```

I suppose that the problem is specific to the resolution of the parameters passed to a macro under a given namespace.

rhaschke, 2025-06-04:
I agree that this is an issue. The quick solution (employing the caller's namespace) breaks the test `test_property_resolution_with_namespaced_include`:
https://github.com/ros/xacro/blob/d3265d04a991b4ad8b645022f06326b6fa7b9df3/test/test_xacro.py#L1366-L1380

However, one could argue that this is not a reasonable use case: The property `inner` is actually not defined at the caller's scope.

AntoSave, 2025-06-04:
I do agree with your point. I would also find it more intuitive if the output of 
```xml
<xacro:B.bar arg="${var}"/>
```
was
```xml
<a arg="main" ext="main" var="2"/>
```
whereas I would expect that
```xml
<xacro:B.bar arg="${B.var}"/>
```
would yeld 
```xml
<a arg="2" ext="main" var="2"/>
```

commit ea664ac4 (2025-08-19): Resolve macro arguments within caller's scope (#373)

Fixes #366.