ros/xacro #368: PyPI xacro version is outdated

adamheins, 2025-07-26:
The current version of xacro in this repo is 1.14.20, while the [PyPI package](https://pypi.org/project/xacro/) is stuck at 1.13.3 from 7 years ago. It would be nice if the PyPI package were kept in sync with the development here to facilitate easier use in independent Python projects (not necessarily coupled to ROS).

traversaro, 2025-07-28:
At a first glance, it does not seem that the PyPI package of xacro was uploaded by the xacro mantainers, see https://pypi.org/project/xacro/ that was uploaded to https://pypi.org/user/safijari/ . So I guess it could make sense to mention @safijari, assuming that the GitHub user and the PyPI user match? 

Similar issue on the `urdfdom-py` side: https://github.com/ros/urdf_parser_py/issues/85 .

adamheins, 2025-08-05:
Poking @safijari again: are you the owner of the xacro PyPI package?

safijari, 2025-08-05:
Apologies for the delay in responding.

Yes. I own the package. I'm happy to transfer it to another maintainer.

rhaschke, 2025-08-06:
I am happy to upload releases to PyPI, ideally automated via some github action. However, I didn't look into this yet - due to limited time. Any help is welcome! To allow new releases on PyPI, @safijari should add me as a maintainer there, I guess.

adamheins, 2025-08-07:
Seems like there is infrastructure to automatically build and publish releases to PyPI e.g. whenever a tagged commit is pushed; see [here](https://packaging.python.org/en/latest/guides/publishing-package-distribution-releases-using-github-actions-ci-cd-workflows/).